ZSWatch/Watch-DevKit-HW #3: Please fill in total quantity of single PCBs

HUGOW04, 2023-10-23:
[image]

HUGOW04, 2023-10-23:
I can also not upload 4 files just one on the upload .png 
[image]

jakkra, 2023-10-23:
Hi @HUGOW04 the quantity is the amount of PCBs you want PCBWay to assemble, you fill that in.

You can add more files when you have it on the Cart also. I ordered just 2 weeks ago and could upload all necessary files.
[image]

jakkra, 2023-10-23:
The above looks correct, there will always be an audit from pcbway for assembly, you will get a quoted BOM from them with the price for components.

HUGOW04, 2023-10-23:
oh thanks I see but do I need to buy some special display and what bettery and display do you recommend??

jakkra, 2023-10-24:
Yes, here:
https://www.buydisplay.com/240x240-round-ips-tft-lcd-display-1-28-capacitive-touch-circle-screen
https://www.aliexpress.com/item/4000121983257.html?spm=a2g0o.order_list.order_list_main.5.4b3418027XxHjE

closed 2023-10-24 by HUGOW04