ZSWatch/Watch-DevKit-HW #9: 1.8-3.3V level shift

DREAM669, 2023-11-15:
Hello, I have noticed that the chip you are using for the 1.8V-3.3V level shift is TXB0104QRUTRQ1, but you are currently using NTS0104GU12. May I know the reason? Is there any defect in TXB0104?Thanks!

Kampi, 2023-11-15:
Hi @DREAM669,

can you explain it further? We are using `NTS0104GU12` in both the schematics and the BOM (at least for rev 3). In theory, both versions should work. The `TXB0104QRUTRQ1` supports a higher voltage range and costs a little bit more.

DREAM669, 2023-11-15:
Hello, I think so. The first version of the schematic used the TXB0104 module, but the later version was changed to NTS0104. Is TXB0104 unable to adapt to Touch functionality?

jakkra, 2023-11-15:
> Hello, I think so. The first version of the schematic used the TXB0104 module, but the later version was changed to NTS0104. Is TXB0104 unable to adapt to Touch functionality?

Yes if I remember correctly it was touch not working because that logic shifter didn't work well for i2c.

Kampi, 2023-11-15:
Hi @DREAM669 

does it answer your question and can we close this ticket?

DREAM669, 2023-11-16:
Of course, my problem has been resolved. Thank you.

closed 2023-11-16 by DREAM669