ZSWatch/Watch-DevKit-HW #10: Part Substitutions from PCBWAY

taintedkernel, 2023-12-09:
Hello, thank you for all the work on this project.  I'm looking to get an assembled device from PCBWAY and I submitted the files that you've created.

They responded that a few items needed review:
* AT25SL128A-MHE-T to be substituted for AT25SL128A
* RD90_LZH0230529-004_DJ-5PIN_PH2_54 needs an exact part number but I see it's noted as "do not assemble"
* OK-24F024-04 cannot be found and a replacement is needed

Appreciate any help you can provide here.

WenchongWang, 2023-12-09:
这是来自QQ邮箱的假期自动回复邮件。邮件已经收到，我会尽快回复您。

Kampi, 2023-12-09:
Hi @taintedkernel

- AT25SL128A-MHE-T -> Correct
- RD90_LZH0230529-004_DJ-5PIN_PH2_54 -> Not available for regular ordering. @jakkra can provide this part to you and you have to solder it on your own (or send it to PCBWay)
- OK-24F024-04 -> Should be available through PCBWay. @jakkra can you please check the part in the BOM of your last order?

jakkra, 2023-12-10:
@taintedkernel PCBway should be able to source the OK-24F024-04
See quoted BOM 
[image]

taintedkernel, 2023-12-12:
Thank you @Kampi @jakkra, I've passed this along to PCBWay and will await their response.

taintedkernel, 2023-12-15:
I received the same response back from them, they cannot source the part.  I see an item available on Alibaba which looks like a match:
* https://www.alibaba.com/product-detail/OK-24F024-04-Connector-Electronic-Components_1600483272580.html

I have a decent amount of soldering experience - would this component be something I can hand solder myself after the rest of the device is assembled and shipped?  The leads seem fairly well-exposed and from what I can tell via KiCad there looks to be sufficient clearance to get an iron in.  Mainly, I'd want to avoid using a hotplate after assembly of the rest of the device.

jakkra, 2023-12-15:
> I received the same response back from them, they cannot source the part. I see an item available on Alibaba which looks like a match:
>
> * [...]

Very strange. Don't know what else I can do, they have sources it for me successfully on 3 different orders. 
If you didn't already order the display you cab order it together with the connector https://www.buydisplay.com/240x240-round-ips-tft-lcd-display-1-28-capacitive-touch-circle-screen

jakkra, 2023-12-15:
You could try giving them this order number and say in it they managed to source the components
 T-X24W564276A

[image]

taintedkernel, 2023-12-15:
Thanks for the info.  I passed along that order number to them and will await their reply.  If they can no longer get it, then I'll go the route of ordering the connector with the display.  I'm fairly confident I'll be able to solder it with an iron or hot air.

I wonder if swapping to a different part might be necessary at some point?  I don't have much experience sourcing connectors, but I've definitely seen similar ones in many devices I've disassembled - it does seem odd there isn't much luck with a match on DigiKey.

Kampi, 2023-12-15:
Hi @taintedkernel,

I think we should qualify an additional variant, maybe from Mouser or DigiKey, as a replacement to prevent additional issues like that in the future.

taintedkernel, 2023-12-18:
@jakkra It seems that order number helped them track the part down, so I've submitted my order and will see how it turns out.  And I certainly agree with you @Kampi that having another variant would be helpful and hopefully mitigate this.

Since I was able to get through this, should I close the issue?  Maybe open a new issue for the part variant idea?  Or keep this one open?

Kampi, 2023-12-18:
@taintedkernel I´ve created a ToDo for this issue, so you can close the the case.

https://github.com/users/jakkra/projects/2?pane=issue&itemId=47711312

The additional variant will be qualified during the work for rev 5 and I then we can add it to rev 4 too.

closed 2023-12-18 by jakkra

jamolnng, 2024-01-09:
> Thanks for the info. I passed along that order number to them and will await their reply. If they can no longer get it, then I'll go the route of ordering the connector with the display. I'm fairly [...]

@taintedkernel were you able to get it sourced? I am having the same issue and am forwarding them that previous order number as well

taintedkernel, 2024-01-12:
> > Thanks for the info. I passed along that order number to them and will await their reply. If they can no longer get it, then I'll go the route of ordering the connector with the display. I'm fairly confident I'll be able to solder it with an iron or hot air.
> > I wonder if swapping to a different part might be necessary at some point? I don't have much experience sourcing connectors, but I've definitely seen similar ones in many devices I've disassembled - it does seem odd there isn't much luck with a match on DigiKey.
> 
> @taintedkernel were you able to get it sourced? I am having the same issue and am forwarding them that previous order number as well. I did get 2 with the display from buydisplay so I could hand solder, I'd just prefer not to

I've had a lot of difficulty with them in general and I'm going the route of hand-soldering.  The agent I communicated with mentioned that it's possible for you to ship them the part and they can assemble it, if you want to go that route.