ZSWatch/Watch-DevKit-HW #11: Minor additions/corrections for PCBWAY assembly

taintedkernel, 2023-12-29:
Hello, I received 2 more questions from PCBWAY regarding assembly and wanted to drop a note here.

First, the BOM has a component marked as R10 listed, but I do not see it on the schematic or PCB - so I'm guessing it was from a previous revision. I told PCBWAY to skip this, let me know if that is incorrect.

And second, a few of the ICs did not have pin 1 marked so they asked me. I looked at the PCB and highlighted the correct pin from what I saw. Attaching a image they provided with my highlights; it looks to be ICs 9 12 14 and 15.

[image]

Kampi, 2023-12-30:
Hi @taintedkernel,

you should also miss `R6` in your BOM. The designators are wrong and `R10` is `R6`. Thanks for the correction. 

You can see the pin 1 markings in `Ref_Top.pdf` in the `docs` folder.

[image]

taintedkernel, 2024-01-05:
Thank you @Kampi - they have not noticed this yet, but I will point it out.

They also pointed out for S1-S4 there is no solder mask opening for holes on top side and holes are covered with solder ink - they are asking if the holes are ok without opening pads on top, it can be reworked by them.  But if pads are necessary they would need a new order.

From what I see on the datasheet, they look to be through-hole and simply mechanical & not electrically connected, so I don't think pads would be necessary on the top?

[image]
[image]

Kampi, 2024-01-05:
Hi @taintedkernel,

you are right. The holes are only for mechanical stability. You can decide if you want a rework or not. I will adjust it in the footprint.

Btw. this is the first time I heard about this error. @jakkra do you had this issue already?

jakkra, 2024-01-05:
@taintedkernel @Kampi I have done 3 orders without any issue, they never noticed this. You can go ahead, for me pcbway have probably soldered manually the bottons through holes afterwards.

taintedkernel, 2024-01-12:
So I've heard back and they are saying the cause was due to a lack of a solder mask opening on the top side. The factory claims the holes are too small to rework and they suggest updating the design, then placing a new order.

Am I missing something? I used the files that are currently in the repo labeled 20231029 and it looks like there is an opening. At this point it's getting very frustrating dealing with PCBWAY and all of these problems.

Kampi, 2024-01-12:
@taintedkernel I´m sorry for this trouble. As @jakkra mentioned we hadn´t had this problem so far and we ordered ~20 or more PCBs there. I checked an empty PCB on my desk and it looks good and there is no soldermask on the holes.

[image]

From my point of view, we have the following options:

- They try to assemble the push buttons and remove the pin for the hole. This might cause some more pressure on the pads
- They ship the buttons unsoldered and you try it on your own
- You have to order new PCBs and I have to check how we can avoid this

What is your preferred way to go? You can also forward your e-mail conversation to me (kontakt@daniel-kampert.de) and I can try to talk with them to find a solution.

taintedkernel, 2024-01-12:
@Kampi I very much appreciate your (and @jakkra's) help on this and my frustration is definitely at PCBWAY!

From what I understand, it seems they have made an error somewhere in the manufacturing since your PCB doesn't have this issue and they were generated from the same source files.  I'm not sure I trust their job enough now for me to be able to fix whatever the issue is.

Assuming this is correct, I'd prefer them to fix their mistake and make a new set of PCBs.  Thank you for offering to jump into the conversation - I will go ahead with this route.

Kampi, 2024-01-12:
@taintedkernel I´m not sure if it´s 100% an issue of PCBWay alone. I´ve figured out that the pad of the switch has no solder mask on the bottom (masked out), but the hole has solder mask (not masked out). It looks like the solder mask goes into the via. I adjusted the footprint to fix this issue now. The question is why hasn´t it happened in the previous orders. Maybe they changed the mask or the process. A good indicator for this is this marked via.

[image]

Here you can see that this via is covered by a bit solder mask, but not total. So it looks like the process isn´t very clean because otherwise all holes must be covered in the same way ("wrong" data -> same result on every location). But we have different results.

We can try to argue in this way with PCBWay and see what happens.

closed 2024-01-21 by jakkra