ZSWatch/Watch-DevKit-HW #23: Removing pull-up and pull-down resistors for vibration motor driver?

Kampi, 2024-01-29:
- Checked the datasheet but don´t find any need for them
- Checking with Jakob if we can move the resistors into the MCU

jakkra, 2024-01-29:
If I remember correctly I added this to make sure the motor was disabled when the voltage regulator removed power to the nRF, hence pin would float.

Looking a bit more into the nPM1300 I think we always should use `VBUSOUT` instead of VBAT (connected directly to battery). This way it will be impossible for any component to pull current after the nPM1300 under voltage hits.

So I think we should use VBUS-MCU (VBUSOUT) as input to the vibration motor.

jakkra, 2024-01-29:
Update this comment to say 50mA instead.
[image]

Kampi, 2024-01-30:
> Update this comment to say 50mA instead. [...]

Good point. 100 mA are related to the LOADSW feature.

jakkra, 2024-01-30:
> > Update this comment to say 50mA instead. ![image](https://private-user-images.githubusercontent.com/4318648/300636546-78a597b8-6134-45eb-b1b7-ca479c9b87af.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY1OTg2MzEsIm5iZiI6MTcwNjU5ODMzMSwicGF0aCI6Ii80MzE4NjQ4LzMwMDYzNjU0Ni03OGE1OTdiOC02MTM0LTQ1ZWItYjFiNy1jYTQ3OWM5Yjg3YWYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMzBUMDcwNTMxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MzJkMjg1YWQ2MmVkZGQwY2M4ZDljOGY1NmFkOGE1OWMwOTVhOTNkZGM5NTNhZDFmNzlkNTFlNjZhOWM4OTc5YiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.yKqRag6sNuBcXO_IWfPdxZVKImVbxLVDWCoqTq7m4UU)
> 
> Good point. 100 mA are related to the LOADSW feature.
> 
> @jakkra please use the right issue next time :D

Oops thought I was in the PR, too many tabs open...

Kampi, 2024-01-30:
@jakkra I close this. I moved your idea with VBAT into a separate issue and link the screenshot to the nPM1300 issue.

closed 2024-01-30 by Kampi