ZSWatch/Watch-DevKit-HW #29: 3D file

DREAM669, 2024-02-29:
Hi,@jakkra 
I have found button.inner.stl and button.inner.angled.stl in your CAD file. May I ask if there is any difference between these two?

WenchongWang, 2024-02-29:
这是来自QQ邮箱的假期自动回复邮件。邮件已经收到，我会尽快回复您。

jakkra, 2024-02-29:
Here it's marked which ones are the angles ones https://github.com/jakkra/ZSWatch/wiki/Case,-3D-printing-and-assembling#buttons

DREAM669, 2024-02-29:
Okay, I understand. Is there any difference between these two sizes?

jakkra, 2024-02-29:
@DREAM669 Yes you need two of each and they need to sit at the location specified in this image:
[image]

DREAM669, 2024-03-04:
@jakkra ,Hi,Can you use a vernier caliper to measure the thickness of the battery and let me know?

jakkra, 2024-03-04:
> @jakkra ,Hi,Can you use a vernier caliper to measure the thickness of the battery and let me know?

3.7mm

DREAM669, 2024-03-05:
Hi,@jakkra 
May I ask what material is used for 3D printed watches on your end? What kind of glue is used to stick the buttons? I'm worried that paying taxes may not be able to stick it firmly. Thank you for your reply!

jakkra, 2024-03-05:
> Hi,@jakkra May I ask what material is used for 3D printed watches ? What kind of glue is used to stick the buttons? I'm worried that paying taxes may not be able to stick it firmly. Thank you for your reply!

I have used PLA. For the buttons the outer cap is optional, so you can skip gluing it alos. But for those I use super glue, for example: https://www.loctiteproducts.com/en/products/fix/super-glue/loctite_super_gluelongneckbottle.html

DREAM669, 2024-03-05:
Do you mean to use the one inside directly? Will it not move without an outer cover, and even fall into the watch?

closed 2024-03-16 by jakkra