ZSWatch/Watch-DevKit-HW #31: Bring back the heart rate sensor

Kampi, 2024-03-01:
- [MAXM86161EFD+T](https://www.mouser.de/ProductDetail/Analog-Devices-Maxim-Integrated/MAXM86161EFD%2bT?qs=XeJtXLiO41T6hWnBlrpJ0w%3D%3D)
- [MAX32664GTGC+](https://www.mouser.de/ProductDetail/Analog-Devices-Maxim-Integrated/MAX32664GTGC%2b?qs=vmHwEFxEFR%252B4L9bBONurNg%3D%3D)
- Normal PCB with flex cable (i. e. https://www.amazon.de/Aexit-Steckverbinder-f%C3%BCr-St%C3%BCck-Bottom/dp/B07C96ZNF2)
- Or flex PCB with integrated cable?

blubbsy, 2024-10-22:
if using the MAX3266...-Hub, shouldnt the accelerotmeter then also a accelerometer being connected to it as it is recommended in the datasheet? 

[image]

here they mention the ROHM KX-11, but also the Bosch BMI323 and LIS2DS12 appear to be possible (I found examples with the old BMI160), but i wonder if the available BMI270 could be used for it.

Actually it hough maybe the MAX86174A could be an option as a HR-Sensor, because it has much more features, but i think it will get much more complicated (and bigger) to do this.

Kampi, 2024-10-22:
AFAIK it´s possible to use an external Acc and send the data to the sensor hub. Jakob or I have looked it up. But the whole HR concept isn´t final yet. It´s more a collection of ideas and depending on the date when we start with the integration we can rethink the component selection and maybe use the bigger one and maybe also a second Acc. Currently I would prefer the LIS2DS12 because of the price :)

jakkra, 2024-10-22:
> AFAIK it´s possible to use an external Acc and send the data to the sensor hub. Jakob or I have looked it up. But the whole HR concept isn´t final yet. It´s more a collection of ideas and depending [...]

You can feed it external Accelerometer data, I looked it up a while ago. So no Accelerometer directly connected needed.

blubbsy, 2024-10-22:
i would rather try to avoid having 2 acc :) i tried to understand in 15 minutes a little bit of how the stuff works, but well... not that easy. i think rohm also has some HR sensors, but what you selected seemed to me much better/easier. 

> You can feed it external Accelerometer data, I looked it up a while ago. So no Accelerometer directly connected needed.
perfect! 

What was the idea behind the FlexPCB? To position the LEDs/Sensor on the right height next to the charger/programmer-interface?
I quickly added it to the assembly and measured the height difference between these two and maybe it is sufficient to put it on a 2nd PCB for the right height? would just require few solder pads and make the assembly easier. 

[image]

Kampi, 2024-10-22:
@blubbsy I accidentally edit your message instead of writing a new -.-

> What was the idea behind the FlexPCB? To position the LEDs/Sensor on the right height next to the charger/programmer-interface?

We have to position the HR sensor so he can touch the skin. The current setup is a PCB with a dock connector. The dock connector adds 10 (?) millimeters of height between the skin and the PCB bottom side. So we have the option to shrink down the dock connector or add a second PCB to handle the distance. The first idea was to use a flex PCB. The current (and latest) idea is to add a second PCB and use small PCB connectors to attach it to the mainboard.

[image]

blubbsy, 2024-10-22:
okay, this works too. i dislike connectors, because i got educated at work that connectors are expensive ;) 

yeah, this could owrk too for the few cents, but then the HR has to be well positioned, so that it does not make any force on the connector or looses contact. And thinking about how to protect against sweat. Nothing that cannot be solved. 

Though i liked my idea and the Bottom is basically empty. 

is the plan to have 1 or multiple of the HR sensors integrated?

Kampi, 2024-10-22:
The plan is to add one HR sensor only. And I think we must use two connectors to add more stability into the construction. The connectors are cheap and for now it's the best solution because we also need a small height. But anyway we had to brainstorm the idea as soon as we continue the work on it. It's paused for now.