ZSWatch/Watch-DevKit-HW #66: NORA-B10 or NORA-B12

blubbsy, 2024-09-18:
Hi,

is there a reason why not using the NORA-B12 with Bluetooth v5.3 instead of 5.2? As far as i see the price are the same (actually on mouser the B12 is 40 cent cheaper). 5.3 should use less energy, which would benefit the Watch. 

The only reason why not using the B12 is that they are running with a max. Voltage of 3.6V instead of 5.5, but as most other ICs are running on 3.6V i don't see the problem.

this is just on first sight, without checking if on SW side the registers changed or whatever... still trying to find some time to take a look on the schematics :/

WenchongWang, 2024-09-18:
这是来自QQ邮箱的假期自动回复邮件。邮件已经收到，我会尽快回复您。

jakkra, 2024-09-18:
Actually it's the same chip and the Bluetooth version can be ignored as it's the same HW, could be just documentation issue. Anyway there is no power consumption difference in v5.2 vs v5.3. The real difference is actually that NORA-B12 have an internal PA (to amplify TX power), but this uses some pins from the nRF5340 which may limit us as we may want those pins.

closed 2024-09-18 by jakkra