ZSWatch/Watch-DevKit-HW #76: DRC Check (PCBWay)

blubbsy, 2024-10-01:
Hi,

Were any design rules imported from PCBWay? (Source: https://www.pcbway.com/pcb_prototype/PCB_Design_Rule_Check.html)
I imported these infomration and i get some errors (in total 411) , here some examples: 

[image]
[image]

I mean the differences are not that big and probably not very severe, but still it is probably not the best to violate the recommendation of the manufacturer. Probably most of these can be easily fixed and i would actually thinking of doing it the next few days/weeks. As probably half of these are the via hole sizes, thats not difficult to fix and many clearance errors can be fixed by moving the vias sligthly (not the ones right below the ICs).

I'm not sure if the minimum trace width should be fixed at all, because probably most will throw a DRC error anyway, because they have to be narrowed for the in-between tracing. 

@Kampi offtopic question: is it okay for you if we write in german? ;) 

LG
Adrian

Kampi, 2024-10-01:
Hi @blubbsy,

let´s use English please if someone else has the same question :)

I don´t use the values from the PCB design rule check page. Instead, I use the values from the PCB Capabilities ([here](https://www.pcbway.com/capabilities.html)). 

I´m not sure where the values from PCB come from. You can compare both values here (the bottom is ours)

[image]

- uVias doesn´t care
- Hole size is set to the value from the configuration mask (0.25 is the smallest hole without switching to "Advanced")
- Min. spacing is also taken from the configuration mask (5 mil is the smallest without switching to "Advanced")
- Via diameter is calculated from 0.25 hole size + 0.15 mm annular ring from the capabilities page (plus 0.05 mm extra)
- Restring is the same in both
- Spacing is set to 0.15 mm (6 mil) to not switch to "Advanced"

[image]

Kampi, 2024-10-01:
Btw. you can check the PCBWay screenshot in the `production` directory from the CI/CD pipeline for the settings. We order the PCBs with these settings every time and they pass the check most of the time. PCBWay is complaining sometimes about the footprint of a sensor. But this is the recommended footprint from the manufacturer so it´s fine.

blubbsy, 2024-10-02:
> By the way, you can check the PCBWay screenshot in the production directory from the CI/CD pipeline for the settings.

True! totally forgot about this, when i checked it yesterday 🤦 

I uploaded it to PCBWAy and there werent any complains at all, except about one solder mask bridge, which.... well... i dont care much:
[image]
[image]

Are you using the PCBA service of PCBWay or are you soldering everything on your own? i'm thinking to order some PCBs and still checking if i want to assembly everything on my own or try out PCBWays service. Depends how lazy i'm...

Kampi, 2024-10-02:
This IC is always critical but the footprint is the recommended footprint from the manufacturer. So you have a distance of 0.18 but you need 0.19. So it´s not a big deal.

[image]

We use the PCBA service from PCBWay all the time and we don´t have any issues with this IC AFAIK. And we order the PCBs in blue or greeen so we can use the 0.19 mm distance.

Anyways the solder mask bridge is basically solder mask between two pads from the IC. It´s not important to have solder mask between this pins if your soldering process is good because the mask stops the solder paste bridging two pins

blubbsy, 2024-10-06:
so here a summary: 

### Cosmetics 
So most of the differences are cosmetic or because of the unit systems, see here the more cosmetic differences: 

| Name                               | ZSWatch   | PCBWay                | Severity | Comment                       |
|------------------------------------|-----------|-----------------------|----------|-------------------------------|
| Solder Mask minimum web width      | 0 [mm]    | 0.1016 [mm] / 4 [mil] | low      |                               |
| Text: Silk Layer Thickness         | 0.12 [mm] | 0.15                  | low      | Worst case: text is not nice. |
| Text: Copper Layers Text Width     | 0.8 [mm]  | 1 [mm]                | low      | Worst Case: text is not nice. |
| Text: Copper Layers Text Height    | 0.8 [mm]  | 1 [mm]                | low      | Worst Case: text is not nice. |
| Text: Copper Layers Text Thickness | 0.12 [mm] | 1 [mm]                | low      | Worst Case: text is not nice. |
| Text: Fab Layers Text Width        | see above | see above             | low      |                               |
| Text: Fab Layers Text Height       | see above | see above             | low      |                               |
| Text: Layers Text Thickness        | see above | see above             | low      |                               |

### Vias
But what i found is that it would be possible to reduce the via sizes actually, because the min. drill size that PCBWay still puts as "normal" would be 0.2mm instead of 0.25mm. This could maybe help a little bit with routing on this narrow space, though i dont understand why in their KiCAD setup they are also setting 0.254mm:

| Name                          | ZSWatch | PCBWay | Severity | Comment |
|-------------------------------|---------|--------|----------|---------|
| Track Width                   | 0.2  [mm]   | 0.1524 [mm] |          |         |
| Clearance                     | 0.125 [mm]  | 0.15 [mm]  |          |         |
| Via Size                      | 0.45 [mm]   | 0.452 [mm] |          |         |
| Via Hole                      | 0.25 [mm]   | 0.254 [mm] |          |         |
| Differential pair track width | 0.2 [mm]    | 0.1524 [mm] |          |         |
| Differential pair track gap   | 0.25 [mm]   | 0.1524 [mm] |          |         |

### Custom DRCs

What i also curious is that PCBWay actually uses some custom DRC rules in their example and if i understand this correctly (never used these) **they reduce the track width and clearance to 0.09**. which is maybe also not that bad to know. :

``(version 1)
#PCBWay Custom DRC for Kicad 7

# ----------------------------------- Minimum trace width and spacing (PICK ONE) --------------------

# 2oz copper
#(rule "Minimum Trace Width and Spacing (outer layer)"
#(constraint track_width (min 0.1524mm))
#(constraint clearance (min 0.1778mm))
#(layer outer)
#(condition "A.Type == 'track'"))

#(rule "Minimum Trace Width and Spacing (innner layer)"
#(constraint track_width (min 0.1524mm))
#(constraint clearance (min 0.1778mm))
#(layer inner)
#(condition "A.Type == 'track'"))

# 2-layer, 1oz copper
(rule "Minimum Trace Width and Spacing (outer layer)"
(constraint track_width (min 0.127mm))
(constraint clearance (min 0.127mm))
(layer outer)
(condition "A.Type == 'track'"))

(rule "Minimum Trace Width and Spacing (inner layer)"
(constraint track_width (min 0.1mm))
(constraint clearance (min 0.1mm))
(layer inner)
(condition "A.Type == 'track'"))

# 4-layer , 1oz and 0.5oz copper
#(rule "Minimum Trace Width and Spacing (outer layer)"
#(constraint track_width (min 0.09mm))
#(constraint clearance (min 0.09mm))
#(layer outer)
#(condition "A.Type == 'track'"))

[... 109 lines cut ...]
# ----------------------------------- Board Outlines (PICK ONE) -------------------------------------
#Default Routed Edge Clearance
(rule "Trace to Outline"
(constraint edge_clearance (min 0.3mm))
(condition "A.Type == 'track'"))

#Special Clearance for V-Score Edges
#(rule "Trace to V-Cut"
#(constraint edge_clearance (min 0.4mm))
#(condition "A.Type == 'track'"))

# ----------------------------------- silkscreen --------------------------
(rule "Minimum Text"
(constraint text_thickness (min 0.15mm))
(constraint text_height (min 0.8mm))
(layer "?.Silkscreen"))

(rule "Pad to Silkscreen"
(constraint silk_clearance (min 0.15mm))
(layer outer)
(condition "A.Type == 'pad' && (B.Type == 'text' || B.Type == 'graphic')"))
``

### PCB Stack
this fits quite well, not sure how much you actually defined it for your own. But what i wonder, how well did your PCB Stack actually fit at the end with the total height of the received PCBs? Worst case this makes the difference if the housing closes or not (sorry i'm used that <1mm differences are important...)

Kampi, 2024-10-06:
The definitions for the pad stacks are provided by PCBway too. AFAIK we don't had any issues with the total height. We order the PCB as 1 mm board and the height will normally differ by something like 0.1 mm.

blubbsy, 2024-10-06:
> The clearance topic is a bit odd because PCBWay says they can produce 0.1 mm with the normal process but they recommend 0.15 mm. So we use the middle of both :)

i also havent understood why the clearance should differ between a 2 and 4 layer PCB. As fasr as I know the process steps are the same, just putting more layers on top. 

> I guess PCBWay hasn't updated their design rules for KiCad. So maybe they are outdated. I guess the rules will fit for the most projects and if they don't fit the people will be experienced enough to adjust it.

I was thinking about this too, but they were done for KiCAD 7, which was release february 2023, so it cant be that outdated at all. And following to the date of this article https://www.pcbway.com/helpcenter/design_instruction/PCBWay_Custom_Design_Rules.html the data is from May 2024. I learned through this how to use the custom DRCs, never looked at it, never though about it. quite cool actually. 
I will just write them and ask :)

Kampi, 2024-10-06:
> 
> i also havent understood why the clearance should differ between a 2 and 4 layer PCB. As fasr as I know the process steps are the same, just putting more layers on top.
> 

Maybe they use a different machine for 4-layer PCBs. It´s quite common for manufacturers to have different machines for PCBs.

> I was thinking about this too, but they were done for KiCAD 7, which was release february 2023, so it cant be that outdated at all. And following to the date of this article [...]

Let me know what the answer was :)

blubbsy, 2024-10-07:
basically they just referred me to: https://www.pcbway.com/multi-layer-laminated-structure.html, though that the information is not consistent, because following to their website, the L1 and L4 copper planes are 0.0175mm thick, instead of 0.035 (which you have in the PCB and what they put in their DRC template):
[image]

I'm actually not sure if they look at it at the end or when selecting 4 layers and 1mm thickness, automatically leads to the stackup. 

i played with the pcbway cost calculator and i think it is possible to reduce the via hole size from 0.25mm to 0.2mm, without additional cost (at least it does not increase for me). **As you have still plenty of space on the PCB it does not help much, but maybe in case it is planned or necessary to be more compact it helps. Should i commit the change (0.2 hole + 0.15mm via diameter == 0.35)?** 

The track width could be reduced to 4mil on the inner layers (i assume that normal process does not increase the price): 
[image]
but i think the DRC checks do not differ between outer and inner layers in KiCAD. SFor the sake of simplicity and to have the proper checks i would not touch this at all.

Kampi, 2024-10-22:
@blubbsy is this issue solved?

blubbsy, 2024-10-22:
i think yes. i will just one day create a PR with the corrected stackup values from PCBway and then it is up to you to accept it or not. it's just changing few values. 

To Summerize:

* Plane thicknesses adapting to pcbway values (i think worst case it influences just the step export).
* Changing to smaller vias. i dont think you will change the vias if not necessary, but when it may be necessary nothing starts to cry. 
* WONT change Inner trace width to 4mil, because i dont know how to define it in KiCAD.

blubbsy, 2024-10-22:
i mean i reduced it and added the via size, but i dont see the necessity to update them all if there is no need for it (at the moment). i hadnt the feeling that you lack of space on the PCB. 

i mean i just quickly changed all the vias, refilled the areas and uploaded it to pcbway and the price did not change at all. So the information on their capability-site seems to outdated? Okay, they also say via sizes should be the same, which i think is just wrong.

[image]
[image]

the price estimation does not change at all. so from my point of view this is then OK.

closed 2025-05-05 by Kampi