ZSWatch/Watch-DevKit-HW #77: External KiCAD Library

blubbsy, 2024-10-07:
Hi,

you built this really amazing library for your purposes ! cant imagine how much work this actually is to maintain (1GB?!). 
but what i wonder if for the projects this is the right approach? shouldnt ZSWatch-HW use it's own project specific library instead of your universal one? I understand why you do it, because it is a lot of work to maintain it.
i would maybe dedicate some time to check if i could write a script (a quite dumb though and i have to check how to do it) that automatically extracts the relevant components from your library to a project library. 

At the same time i would check if there are some other components that could need an update of the paths, because i saw that the LIS2MDLTR is actually still referring to the KiCAD6 path. Dont know why it actually works, because i have a clear kicad installation and kicad6 is not in the paths anymore, but okay...

This is cosmetics, so definitely nothing i want to bother you much. Personally i just downloaded your library and added the path to KiCAD... 

BR
Adrian

Kampi, 2024-10-08:
Hi Adrian,
I also thought about it last week because it makes versioning easier and may avoid some issues, e.g., fetching a specific commit of the watch hardware and using the latest library version, which changes some footprints. This can lead to potential bugs in the hw when a user updates the footprints. Feel free to extract the specific parts and copy them in a' library directory in the project root. Then you can add them as project-specific components and add them to Git. Next, you have to change the CI/CD pipeline.

blubbsy, 2024-10-08:
> I also thought about it last week because it makes versioning easier and may avoid some issues
totally true, don't know why i haven't though about it. Also it is then possible to make releases, that fit together with the SW part. 
Do you want to detach it entirely from your library and maintain a specific library? my idea would have been to write a python script that copies from your library the used components and copies them into the ZSWatch-Project/library folder.

to your ToDo list: sounds good! Dont know when i will do it, because i'm travelling for work to Munich tomorrow. But i will try my best ;-) 

> Discuss if we want to do it for all parts (also, the KiCad parts) or the custom parts only. Doing the same with the KiCad parts might avoid issues when upgrading KiCad

Why the KiCAD parts? i think these are only standard packages (diodes, capacitors, resistors etc.) and i wouldn't expect any changes. If i'm not mistaken these are actually generated through a python script.

Kampi, 2024-10-08:
Take your time. We are not in a rush with this topic :)

Why the KiCad parts:
Only for completion (I. e. when modifing a resistor to remove the silkscreen). We can use the way that we do not change standard components and if we do they should be placed in the library or we copy all components and we add the standard components that currently exist in the layout and then you can change them later on.

Kampi, 2024-10-22:
@blubbsy we must check how we rebase your changes to my 2.0.6 branch. I suggest we try it with [CadLab](https://cadlab.io/) as soon as you have finished your work. It´s also a good check for a new workflow to improve the work as a team on KiCad projects.

blubbsy, 2024-10-22:
> we must check how we rebase your changes to my 2.0.6 branch.

no idea 🤦

> I suggest we try it with [CadLab](https://cadlab.io/) as soon as you have finished your work.

just created already the account for it. Actually mostly the proper check if still everything is correct is the main issue, so CadLab is exactly the right solution for it.

closed 2025-01-02 by Kampi