ZSWatch/Watch-DevKit-HW #80: Understanding need for 3V3/VDD-Display

blubbsy, 2024-10-19:
Hi, 

What is the reason to supply the display with 3.3V instead of 3.0V? If i understand the datasheet of the display correctly, then the typical supply should be 2.8V for the supply and 3.0V for the backlight.

i summed up all the currents from the sensors, mcu, flash etc. and i calculated a total peak current of 75mA (rounded up), while the output could handle 200mA. In normal operating the value should be much smaller, because it is already with flash writing, full display, vibration motor, MCU transmitting etc. 

The reason why I looked at it is, that i think the power supply would be more efficient when only using the 3V instead off additionally using the 3V3 from the LDO1. I first though the LDO1 is used for some software control over the display, but then i saw the backlight driver that actually does it... 

Am I missing something or is there a SW reason to use the LDO1?

BR
Adrian

Kampi, 2024-10-19:
I guess it´s a relict from the previous versions. I haven´t checked it in the past. The separate power supply has the advantage of cutting off the display power completely. If we use one power supply for all we can not turn off the display, instead it´s only possible to reduce the backlight.

Maybe we can use the load switch for it?

@jakkra what was the reason for using 3V3 for the display? Do you know it?

blubbsy, 2024-10-19:
I think i know from where this comes. The datasheet of the display itself does not say anywhere 3.3V as recommended supply voltage, neither does the controller Datasheet (https://files.waveshare.com/upload/5/5e/GC9A01A.pdf) say it. but at many places in the web 3.3V are stated as nominal supply voltage, which i cant really follow why this is the case: 

[image]
[image]
[image]

> If we use one power supply for all we can not turn off the display, instead it´s only possible to reduce the backlight.
this is what i also though, but now that i scrolled through the controller datasheet:
[image]

> Maybe we can use the load switch for it?

Something to make you laught: 
[image]

Kampi, 2024-10-19:
Okay, so removing the backlight supply doesn´t work. If @jakkra hasn´t any argument against it I would suggest this as an idea for Rev 2.6 (at least for testing).

jakkra, 2024-10-19:
Reason it's at 3V3 is that I just picked it as I consider it a common voltage and within the range. Before the nPM1300 there was less flexibility for voltage levels. With v5 we got requirement to have 3V0, and I think we never thought of changing display VBUS to 3V0 also. BUT it's very good to have display on a seperate power supply as it pulls quite a bit of power even in sleep mode (perhaps putting in reset is ok, never tried) so it's simple to just cut the power and touch and display are consuming 0.

blubbsy, 2024-10-19:
> BUT it's very good to have display on a seperate power supply as it pulls quite a bit of power even in sleep mode (perhaps putting in reset is ok, never tried)

Yes, but if the nPM1300 can provide so much more I don't see any problem. And a switch can be also located on the 3V line exclusively the display if necessary. 

Ultimately maybe it can be tested how much the display consumes in sleep to see if it is worth? I expect the buck to be more efficient than the LDO1, but has to be checked.

> Okay, so removing the backlight supply doesn´t work.

I think they are overdramatic 😅

Kampi, 2024-10-20:
Probably we can use an additional pnp transistor to switch the LCD and use own power supply for all?

blubbsy, 2024-10-20:
Someone had to stand up early today ;) 

You mean something like this? 
[image]
The PNP would need to have a lower voltage drop than the LDO (200mOhm), otherwise it would not save energy at all. 

i'm not sure if this somehow benefits us at all. I mean: is the PNP more efficient than using the nPM1300 LDOs? In a nutshell the discussion is about how to make it more energy saving right? 

Initially i only though that the Buck would be more efficient than the LDO. And well, slightly the board layout would be more simply as you have to route one trace less. 

Maybe it is something to postpone and simpyl keeping it in mind for future ideas/optimizations, until some question are answered regarding the energy consumption. If we realize that it is okay to use only the 3V also for the display, we could check if it makes sense to change the sensors to 3V (or whatever voltage level makes sense. they are all already rated for >3V) too and use a smaller PMIC (e.g. nPM1100 --> ~30% cheaper / 2€) as the additional voltage level wouldnt be needed anymore.

Kampi, 2024-10-20:
A DC/DC converter is always more efficient than an LDO because of the way the energy is converted. The LDO converts a voltage difference into heat so we use the LDO to generate 3.3 V out of 4.2 V we waste 0.9 V * 75 mA (just a random value as display power) 0.0675 W. If we use the 3 V from the DC/DC we don´t have this issue but we have the problem that the display can not be disabled alone because all peripherals are powered by 3 V. This is where the PNP comes into play. With this PNP as a simple switch we can turn on and off the display. Using a `DMG1013UW-7` (like for the microphone) means an additional 1.5 Ohms resistor (worst case) which means 1.5 Ohms * (75 mA)^2 or 0,0084375 W = 12.5 % in comparison to the LDO.

The next advantage would be that the watch can be used a bit longer because currently, the display doesn´t work at a battery voltage less than 3.3 V (VOut LDO) + 40 mV (Droput of the LDO).

The reason why we switch from the LDOs to the nPM is the optimized bundle of MCU and PMIC together with an optimized battery charging and a more efficient power supply (DC/DC instead of LDO). The nPM1100 is IMO not an option because we want to use the ship mode of the nPM1300 to shutdown the watch when not needed and the "real" hardware reset of the nPM1300 ist also a big bonus

jakkra, 2024-10-20:
IMO we can keep this in mind for future revision, but not worth doing now current setup is fine and power consumption wise, shoving off a few uA doesn't do much if you use the watch regularly anyway. I however think the features of nPM1300 are not something I want to trade away also.

Kampi, 2024-10-20:
@blubbsy feel free to add this idea as a ticket for rev 2.6 or later :)

There is a lot to do with the current revision (like reducing the current consumption :P).

blubbsy, 2024-10-20:
> I however think the features of nPM1300 are not something I want to trade away also.

Totally legit point. As I dont know the SW side so far, i have no idea which features are being used at all or what capabilites the different nPMxxxx have. 

> @blubbsy feel free to add this idea as a ticket for rev 2.6 or later :)

I cant move/edit this ticket to add it to the milestone. i would like to keep the discussion for documental reasons when someone is starting to work on it.  Who knows when this is being worked on or if other things change (new sensors, displays etc.), but as we had good clarifactions here it would be a shame to have it distributed through multiple discussions.
A new issue would be possible to be assigned to the milestone by me 🤷‍♂️.

Kampi, 2024-10-20:
I can edit it later

closed 2025-04-08 by Kampi