ZSWatch/Watch-DevKit-HW #94: pcbway issue

ravi145258, 2024-11-29:
pcbway opening showing issue for purchasing pcb

jakkra, 2024-11-29:
Not enough information.

ravi145258, 2024-11-29:
https://www.pcbway.com/project/shareproject/ZSWatch_Open_Source_Smartwatch_4ca3bcf6.html
this getting error please share where to purchase pcb

jakkra, 2024-11-29:
> https://www.pcbway.com/project/shareproject/ZSWatch_Open_Source_Smartwatch_4ca3bcf6.html this getting error please share where to purchase pcb

It's been removed temporarily as a new version (v5) will soon be uploaded to PCBWay. So I suggest waiting for that.

ravi145258, 2024-11-29:
thank your verymuch jakkra is there any option to change pcb layout to square like apple watch

jakkra, 2024-11-29:
> thank your verymuch jakkra is there any option to change pcb layout to square like apple watch

If you know KiCad you can do whatever, all source is in this repo.

closed 2025-01-02 by Kampi